CubePilot/qgroundcontrol-herelink #1: Solex Take Picture no triggering Shutter in Autopilot

MadRC, 2020-04-20:
## Expected Behaviour

Take Picture command in button programming should trigger Shutter command in Ardupilot 

## Current Behaviour

When configuring a programmable button to 'Take Picture' in Solex this is not triggering camera shutter action in Ardupilot.  

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Configure button to 'Take Picture' in Solex button mapping 
2. Configure Servo output to Camera Shutter 
3. Test button. 

## System Information

Herelink Beta 

## Detailed Description
The solex take picture command does not trigger the shutter command in Ardupilot, users can map Servo 7 to achieve the same outcome however the 'Take Picture' command is the logical way to set this up.

kellyschrock, 2020-04-29:
Just for clarification: The expectation here is to send a "take picture" command to the vehicle when the shutter button is pressed _on the screen_ (not a hardware button), correct? In that case, I have Solex sending `COMMAND_LONG/DIGICAM_CONTROL` to the sysid of the air unit and compid `CAMERA` trigger it. Is that what we want? I don't have a camera that responds to shutter commands, so can't test it directly here.

MadRC, 2020-05-04:
Ok having looked at this I have an idea, as standard I don’t see this being able to to trigger anything but a fully Mavlink gimbal like the Solo one. My idea is based on what we did with the recording. 

Would it be possible to have default action of the camera still button take a screen shot or image of the video feed and save that locally.  Then in settings like with video have the option of choosing to send the Mavlink command.  

Hope this makes sense.

kellyschrock, 2020-05-05:
Hey @MadRC That's actually what the still button does when you click it and "Air-unit Record" is turned off. Currently the settings are like this:

**Air-Unit Record ON**
- Video record: Send `COMMAND_LONG/VIDEO_START_CAPTURE` (which apparently works, when the air unit is set up to handle video recording output)
- Still photo: Send `COMMAND_LONG/DO_DIGICAM_CONTROL` (shot command)

**Air-Unit Record OFF**
- Video record: Record local video to the Android device in `/sdcard/Movies`
- Still photo: Take a snapshot of the video screen in a .jpg file and save in `/sdcard/Pictures`