CubePilot/qgroundcontrol-herelink #11: QGC video streaming problem with a custom build

hugheslavigne, 2020-05-05:
I'm currently building a payload for environmental scientists and I'm looking to build a custom QGC app for it. At this moment, I'm just trying to build the Herelink QGC app from this repository without any modification.

I'm having difficulties with the video transmission when I build this app without any change to the repository. I'm experiencing video freeze with the custom build that I don't get with the built-in app that comes with the Herelink remote controller

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 16.04.6
- Android SDK version: 26.1.1
- Android NDK version: 15.2.4203891
- Qt Version: 5.11
- Qt Creator version: 4.11.0
- Compiler: arm-linux-androideabi-gcc and arm-linux-androideabi-g++ from the Android NDK version 15

## Log Files
I ran a logcat on the QGC I’ve built from the Herelink repo without any modification and also on the QGC that comes built-in on the Herelink. The main difference that could explain those video freezes are these lines below. It seems that I can’t get OMX started. 

With my built, I noticed this line :
3022 3038 E QGroundControl: /home/eskimo/src/qgroundcontrol-herelink/src/VideoStreaming/VideoReceiver.cc:355 (virtual void VideoReceiver::start()): VideoReceiver::start() failed. Error with gst_element_factory_make(‘amcviddec-omxarmvideov5xxdecoder’)

With the built-in QGC app, I got these lines :
276 2056 I MediaPlayerService: MediaPlayerService::getOMX
2594 2648 I OMXClient: MuxOMX ctor
270 1973 I OMXMaster: makeComponentInstance(OMX.ARM.video.v5xx.decoder) in mediacodec process
270 1973 D MVE_KERN_IF: Firmware version: 0x201 (mve_userspace.c:466)
270 2559 W MVE_OMX_IF: Unsupported extension index. handle=0xef807300, name=OMX.google.android.index.describeColorAspects. (mve_omx_common_component.c:2557)
270 916 W MVE_OMX_IF: Unsupported extension index. handle=0xef807300, name=OMX.google.android.index.describeHDRStaticInfo. (mve_omx_common_component.c:2557)
270 2559 W MVE_OMX_IF: Unsupported extension index. handle=0xef807300, name=OMX.google.android.index.prepareForAdaptivePlayback. (mve_omx_common_component.c:2557)
270 2559 W MVE_OMX_IF: Unsupported extension index. handle=0xef807300, name=OMX.google.android.index.configureVideoTunnelMode. (mve_omx_common_component.c:2557)
270 2559 W MVE_OMX_IF: Unsupported extension index. handle=0xef807300, name=OMX.google.android.index.useAndroidNativeBuffer. (mve_omx_common_component.c:2557)
270 2559 W MVE_OMX_IF: Unsupported extension index. handle=0xef807300, name=OMX.google.android.index.allocateNativeHandle. (mve_omx_common_component.c:2557)

Here are the two complete log I get when I start the Herelink built-in QGC app and the app I built using the repo and Qt:
[built_QGC.log](https://github.com/CubePilot/qgroundcontrol-herelink/files/4580963/built_QGC.log)
[herelink_QGC.log](https://github.com/CubePilot/qgroundcontrol-herelink/files/4580964/herelink_QGC.log)

bugobliterator, 2020-05-07:
@hugheslavigne You need to ensure that you don't have both QGC apps running at the same time. Try to follow the instructions here https://docs.cubepilot.org/user-guides/herelink/oem-image-setup , and place your QGC apk under app/  directory inside oem image. Flash that image, this way your version of QGC will override the built in QGC.

hugheslavigne, 2020-05-07:
Ok, I've got some problem with your OEM image setup, maybe this should be another issue...

At first, I simply created an app/ folder inside the OEM image with my QGC apk in it. It simply didn't work. My app was in the launcher but I couldn't open it (it opens an blank window and its size is 0 MB) and the built-in app was still there. Then, I took a look at the folders on the Herelink with a shell. It seems that the QGC app is located in system/app. So I simply created a system/app folder in my OEM image. With this method, I can't see my app in the launcher, there is only the QGC built-in app. I can confirm that my oem folder with my QGC apk is present in the Herelink after I flashed it using the bootloader. 

Basically, I'm stuck with my original problem, as I am unable to replace the QGC app on the Herelink. I did some tests where I force stop the QGC built-in app to run my custom build, but I'm still experiencing video lagging.

Do you have any idea what might be the problem with my OEM image? Am I doing something wrong?

bugobliterator, 2020-05-08:
@hugheslavigne Can you try building the apk by changing the package name. In QtCreator, under AndroidManifest.xml, simply modify the package name from org.mavlink.qgroundcontrol to say org.mavlink.qgroundcontrol1. Place the apk inside oem/app/ directory, that part was done correctly by you.

hugheslavigne, 2020-05-08:
@bugobliterator I've tried many times to install my QGC through the adb command with various parameters, including changing the app name. Whatever I changed, I keep getting video lagging problems.

I modified the AndroidManifest.xml to change the package name as you suggested. Then, I placed the apk inside oem/app folder. However, Again, I wasn't able to start the application. It seems that the apk isn't install after I flashed it on the device. This is the error log I got on the device : 
721  1249 W WindowManager: addAppToken2: AppWindowToken{329ae14 token=Token{329b067 ActivityRecord{d47cc26 u0 org.mavlink.qgroundcontrol1/org.mavlink.qgroundcontrol.QGCActivity t22}}}
1612  1612 W System  : ClassLoader referenced unknown path: /oem/app/QGroundControl1/lib/arm
1612  1612 E Qt      : Can't create main activity
1612  1612 E Qt      : java.lang.Exception: 
1612  1612 E Qt      : 	at org.qtproject.qt5.android.bindings.QtLoader.startApp(QtLoader.java:572)
1612  1612 E Qt      : 	at org.qtproject.qt5.android.bindings.QtActivityLoader.onCreate(QtActivityLoader.java:183)
1612  1612 E Qt      : 	at org.qtproject.qt5.android.bindings.QtActivity.onCreateHook(QtActivity.java:266)
1612  1612 E Qt      : 	at org.qtproject.qt5.android.bindings.QtActivity.onCreate(QtActivity.java:273)
1612  1612 E Qt      : 	at org.mavlink.qgroundcontrol.QGCActivity.onCreate(QGCActivity.java:203)
1612  1612 E Qt      : 	at android.app.Activity.performCreate(Activity.java:6682)
1612  1612 E Qt      : 	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1118)
1612  1612 E Qt      : 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2619)
1612  1612 E Qt      : 	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2727)
1612  1612 E Qt      : 	at android.app.ActivityThread.-wrap12(ActivityThread.java)
1612  1612 E Qt      : 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1478)
1612  1612 E Qt      : 	at android.os.Handler.dispatchMessage(Handler.java:102)
1612  1612 E Qt      : 	at android.os.Looper.loop(Looper.java:154)
1612  1612 E Qt      : 	at android.app.ActivityThread.main(ActivityThread.java:6121)
1612  1612 E Qt      : 	at java.lang.reflect.Method.invoke(Native Method)
1612  1612 E Qt      : 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:889)
1612  1612 E Qt      : 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:779)

bugobliterator, 2020-05-08:
@hugheslavigne ok, I will have a review on my end and get back to you.

hugheslavigne, 2020-05-22:
@bugobliterator I understand that you might be busy on another matter, but do you have any updates regarding this issue?

bugobliterator, 2020-05-22:
@hugheslavigne Yes I have been quite busy this week. Its on my top priority over this weekend. Apologies for delay.

bugobliterator, 2020-05-23:
@hugheslavigne I have a build which should fix the issue, please ping me on CubePilot forum message, I will share an alpha build that you can test. Thanks for your patience.

closed 2020-06-23 by bugobliterator