CubePilot/qgroundcontrol-herelink #12: Feature Request: Connect computer QGC to herelink QGC

MaxHausmann, 2020-05-08:
Hey there,

is it possible to connect a local version of QGC on my Linux machine via Wifi or USB-Cable to my herelink QGC app. I want to build a groundcontrol station for a research drone which I can set
parameters and control specific things on my drone from the computer while the pilot is flying the
system. 

Comment: I am also a programmer, so maybe I can implement this feature by my own or one of my research colleagues.

Thanks, Max

bugobliterator, 2020-05-09:
@MaxHausmann Follow the steps here https://docs.cubepilot.org/user-guides/herelink/connect-to-mission-planner You can connect to the same UDP port mentioned there for other GCS.

MaxHausmann, 2020-05-09:
Doesn't work for me, neither with herelink hotspot nor with some external wifi.
Are you sure that is possible to connect from QGC to QGC directly? 

In console on Linux machine it says:
``` 
Adding target QHostAddress("192.168.43.1") 14550
[...]
Error writing to QHostAdress("192.168.43.1") 14550
```
The IP 192.168.43.1 should be the herelink remote, because I can ping it from my Linux machine.

meee1, 2020-05-12:
this sounds like something your firewall might be blocking.
by default when you enable the hotspot, the traffic is forwarded to the broadcast address.
 my guess is its being blocked by your firewall

MaxHausmann, 2020-05-12:
Because I can not change any firewall settings on my herelink remote (?), I have to open the port on my linux machine?

meee1, 2020-05-12:
linux machine.
if you can do a wireshark/tcpdump on your wlan interface, to see what traffic you are seeing

MaxHausmann, 2020-05-12:
Okay, thanks I will try that.

MaxHausmann, 2020-05-14:
I opened via ufw the ports 14550 and 14552 in all directions from all IPs.
The error messages are now gone, but I don't really see some connection between the two devices.

I used Wireshark for capturing traffic, in "hotspot-mode" there were no packets at all, in "wifi-mode" my linux machine tries to reach the herelink remote (10.10.10.26 from 10.10.10.20) but the traffic goes only to the remote and no traffic comes back.

After connecting, the herelink QGC-app shows some error message:
```
Detected problems with app native libraries (please consult log for detail): libQGroundControl.so: text relocations
```

In console there are no messages. It is on both sides only claiming about the missing internet connection (in "hotspot-mode")

MaxHausmann, 2020-05-14:
Now I tested it with Mission Planner on a Win 10 machine as well. With hotspot and my wifi network. I also looked at the traffic and the result is the same as on my linux machine.
I thought, maybe I can try to connect my Linux and Windows machine together (with QGC/Mission Planner) but I am not sure if that is possible. But also it didn't work.

meee1, 2020-05-14:
What firmware are you running on herelink?

MaxHausmann, 2020-05-14:
The newest one: BRU01200419
QGC version: herelink-v3.5.6:5daf89741

MaxHausmann, 2020-05-14:
I tried some test to establish a connection to each other on port 13552, the result was that the computer as well as the remote are sending short chunks on this port to each other but they still don't recognize each other.

MaxHausmann, 2020-05-21:
All problems solved. Build up minimal drone setup, and updated all firmwares. 
Now I have not to set any connections, the system does it by itself.
Thank you very much!

closed 2020-05-21 by MaxHausmann

Bulltac, 2020-06-09:
Hey how did you end up solving this issue? Did you have to reflash firmware on the flight controller and the herelink?

MaxHausmann, 2020-06-14:
Hey, in my previous tries I did not use any airborne hardware (flight-controller etc.) so there is no source of mavlink which yields to no connection between all ground devices. The solution is to use
the system with real drone hardware. The update of the firmware was imho not necessary.