CubePilot/qgroundcontrol-herelink #15: There is no way to change to Mode1

bbc25185, 2020-05-15:
## Detailed Description
I am one who uses the remote control normally in Mode1;
After the last update many things have changed and among many others, the possibility of setting the radio mode has disappeared. (Default Mode2)
I tried to set the channels on the flight controller to match my Mode but, probably due to a bug on the FC, it doesn't work (When I exchange channels, the flight controller goes into fail safe and there is no way to get it out except by putting everything back to the original).
The ability to change Modes is now something that all radios should have.

It could be inserted a new page after the calibration menu of the stick, maybe to a menu where you can, too, associate SBus PWM outputs to the various buttons and the scroll wheel. 
(sorry for my poor English)

MadRC, 2020-05-15:
More options will be coming soon for Subs channels in Herelink.  As will RC mode selection and channel reversing ect.  

With regards to mode selection this should always be done in Mavlink buttons and it’s strongly advised that they are not programmed to Sbus due to unpredictable behaviour.  

You need to trouble shoot why that’s not working as it should be fine in both QGC and Solex.

bbc25185, 2020-05-16:
Maybe I explained myself wrong...
As a mode change, I'm not talking about flight modes that you can change using the buttons ...
I speak of the association of the sticks to the main controls:
Mode1: Left stick for pitch and yaw, right stick for throttle and roll!
Mode2: Left stick for throttle and yaw, right stick for pitch and roll!
Mode3: Left stick for pitch and roll, right stick for throttle and yaw!
Mode4: Left stick for throttle and roll, right stick for pitch and yaw!

meee1, 2020-05-16:
there is an update coming very soon that will give you the option on the herelink remote unit to output based on any mode 1-4. just select mode 1 (i am mode 1), and the channel output order will be adjusted accordingly.

closed 2020-06-23 by bugobliterator