CubePilot/qgroundcontrol-herelink #18: Cant compile source code on latest SDK and NDK.

Dhananjay-Khairnar, 2020-12-28:
- Target Compilation : Herelink.
- Host OS : Windows 10
- Platform : QT 5.14
- Android NDK : 21.3.6
- Android SDK : 26.1.1

**Compilation Logs**
`12:12:31: The process "C:\Qt\Qt5.14.1\5.14.1\android\bin\qmake.exe" exited normally.
12:12:31: Starting: "C:\Users\Owner\AppData\Local\Android\Sdk\ndk\21.3.6528147\prebuilt\windows-x86_64\bin\make.exe" -f C:/Dhananjay/build-qgroundcontrol-Android_for_armeabi_v7a_arm64_v8a_x86_x86_64_Clang_Qt_5_14_1_for_Android_540d4d-Debug/Makefile qmake_all
make: Nothing to be done for 'qmake_all'.
12:12:32: The process "C:\Users\Owner\AppData\Local\Android\Sdk\ndk\21.3.6528147\prebuilt\windows-x86_64\bin\make.exe" exited normally.
12:12:32: Starting: "C:\Users\Owner\AppData\Local\Android\Sdk\ndk\21.3.6528147\prebuilt\windows-x86_64\bin\make.exe" -j4
linking debug\libQGroundControl_armeabi-v7a.so 
clang++: error: no such file or directory: '@obj\object_script.debug\libQGroundControl_armeabi-v7a.so.Armeabi-v7a'
make: *** [Makefile:1403: debug\libQGroundControl_armeabi-v7a.so] Error 1`

kasparjj, 2021-05-17:
Hi @Dhananjay-Khairnar 

Did you manage to compile the project? 
I have similar problems with compiling.

Best regards

Dhananjay-Khairnar, 2021-07-05:
I have compiled the code. but its not look the same like QGCS comes pre installed in Herelink. Video streaming is not enabled dont know where to configure it. 

Host OS : Ubuntu 21.04
Android build SDK - 29
Android NDK - 15.2.4203891([android-ndk-r15c-linux-x86_64.zip](https://dl.google.com/android/repository/android-ndk-r15c-linux-x86_64.zip))  (also requires 21.3.6528147 for make)
QT = 5.11.0

Changes required in files:

1. qgroundcontrol/android/AndroidManifest.xml

- `<uses-sdk android:minSdkVersion="21" android:targetSdkVersion="29"/>`
2. qgroundcontrol/src/VideoStreaming/VideoStreaming.pri

- L 78 `GST_ROOT = $$PWD/../../gstreamer-1.0-android-armv7-1.5.2` (set both GST_ROOT path to same)
- L 110 `-lgstvideo-1.0 -lavformat -lavcodec -lavutil -lx264 \#-lavfilter -lswresample \`
- L 114 add belove `LIBS += -L$$PWD/../../lib -lavresample`

closed 2021-07-05 by Dhananjay-Khairnar

HimanshuChowdharyIndia, 2022-03-10:
> I have compiled the code. but its not look the same like QGCS comes pre installed in Herelink. Video streaming is not enabled dont know where to configure it.
>
> Host OS : Ubuntu 21.04 Android build [...]

Hi @Dhananjay-Khairnar ,

I tried on QT 5.11.0 and QT 5.15.2. 
I am using NDK 21.3.6528147. I also tried using NDK android-ndk-r15c.

Actually code gets compiled but I get error while linking .so file. 

herelink did not provided any system environment guidance.

I am trying to build <herelink-v4.0.8> - https://github.com/CubePilot/qgroundcontrol-herelink

Please find below exception I got:

obj/QGeoCodeReplyQGC.o(.ARM.extab+0x0): error: undefined reference to '__gxx_personality_v0'
obj/QGeoCodeReplyQGC.o(.ARM.extab+0xe4): error: undefined reference to '__gxx_personality_v0'
obj/QGeoCodeReplyQGC.o(.ARM.extab+0xfc): error: undefined reference to '__gxx_personality_v0'
obj/QGeoCodeReplyQGC.o(.ARM.extab+0x114): error: undefined reference to '__gxx_personality_v0'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qobject.h:266: error: undefined reference to 'operator new(unsigned int)'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qobject.h:266: error: undefined reference to 'operator delete(void*)'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qobjectdefs_impl.h:415: error: undefined reference to 'operator delete(void*)'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qlist.h:626: error: undefined reference to '__cxa_begin_catch'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qlist.h:628: error: undefined reference to '__cxa_rethrow'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qlist.h:629: error: undefined reference to '__cxa_end_catch'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qlist.h:644: error: undefined reference to '__cxa_begin_catch'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qlist.h:646: error: undefined reference to '__cxa_rethrow'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qlist.h:647: error: undefined reference to '__cxa_end_catch'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qlist.h:807: error: undefined reference to '__cxa_begin_catch'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qlist.h:810: error: undefined reference to '__cxa_rethrow'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qlist.h:811: error: undefined reference to '__cxa_end_catch'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qlist.h:815: error: undefined reference to '__cxa_begin_catch'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qlist.h:820: error: undefined reference to '__cxa_rethrow'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qlist.h:821: error: undefined reference to '__cxa_end_catch'
/Users/himanshu/Anra/qgroundcontrol-herelink/src/QtLocationPlugin/QGeoMapReplyQGC.cpp:85: error: undefined reference to 'operator delete(void*)'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qobject.h:266: error: undefined reference to 'operator new(unsigned int)'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qobject.h:266: error: undefined reference to 'operator delete(void*)'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qobject.h:266: error: undefined reference to 'operator new(unsigned int)'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qobject.h:266: error: undefined reference to 'operator new(unsigned int)'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qobject_impl.h:75: error: undefined reference to '__cxa_guard_acquire'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qobject_impl.h:75: error: undefined reference to '__cxa_guard_release'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qobject_impl.h:75: error: undefined reference to '__cxa_guard_abort'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qobject_impl.h:75: error: undefined reference to '__cxa_guard_acquire'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qobject_impl.h:75: error: undefined reference to '__cxa_guard_release'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qobject_impl.h:75: error: undefined reference to '__cxa_guard_abort'
/Users/himanshu/Anra/qgroundcontrol-herelink/src/Audio/AudioOutput.cc:76: error: undefined reference to '__cxa_guard_acquire'
/Users/himanshu/Anra/qgroundcontrol-herelink/src/Audio/AudioOutput.cc:76: error: undefined reference to '__cxa_guard_release'
/Users/himanshu/Anra/qgroundcontrol-herelink/src/Audio/AudioOutput.cc:76: error: undefined reference to '__cxa_guard_abort'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qobject_impl.h:75: error: undefined reference to '__cxa_guard_acquire'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qobject_impl.h:75: error: undefined reference to '__cxa_guard_release'
/Users/himanshu/Qt/5.15.2/android/include/QtCore/qobject_impl.h:75: error: undefined reference to '__cxa_guard_abort'
/Users/himanshu/Anra/qgroundcontrol-herelink/src/Geo/MGRS.cpp:52: error: undefined reference to '__cxa_allocate_exception'
/Users/himanshu/Anra/qgroundcontrol-herelink/src/Geo/MGRS.cpp:52: error: undefined reference to '__cxa_throw'
/Users/himanshu/Anra/qgroundcontrol-herelink/src/Geo/MGRS.cpp:52: error: undefined reference to '__cxa_free_exception'
/Users/himanshu/Anra/qgroundcontrol-herelink/src/Geo/MGRS.cpp:54: error: undefined reference to '__cxa_allocate_exception'
/Users/himanshu/Anra/qgroundcontrol-herelink/src/Geo/MGRS.cpp:54: error: undefined reference to '__cxa_throw'
/Users/himanshu/Anra/qgroundcontrol-herelink/src/Geo/MGRS.cpp:54: error: undefined reference to '__cxa_free_exception'
[... cut ...]
obj/moc_QGeoCodeReplyQGC.o:moc_QGeoCodeReplyQGC.cpp:vtable for QGeoCodeReplyQGC: error: undefined reference to 'QGeoCodeReplyQGC::~QGeoCodeReplyQGC()'
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Makefile:1660: debug/libQGroundControl_armeabi-v7a.so] Error 1
14:02:58: The process "/Users/himanshu/Library/Android/sdk/ndk/21.3.6528147/prebuilt/darwin-x86_64/bin/make" exited with code 2.
Error while building/deploying project qgroundcontrol (kit: Android Qt 5.15.2 Clang Multi-Abi)
When executing step "Make"
14:02:58: Elapsed time: 13:24.

Best Regards,
Himanshu