CubePilot/qgroundcontrol-herelink #25: Unable to Build : error: cannot find symbol import android.os.storage.StorageVolume;

iamczar, 2021-11-06:
## Expected Behavior
Please tell us what should happen as detailed as possible.

## Current Behavior
unable to build due to error: cannot find symbol import android.os.storage.StorageVolume;
[image]

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. clone qgroundcontrol-herelink and open qgroundcontrol.pro with Qt creator 5.0.3
2. using /java-8-openjdk-amd64 and android-ndk-r15c

[image]

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 20.04
- QGC Version: Current Master

## Compile Output

[image]

iamczar, 2021-11-07:
Hi @DonLakeFlyer , Sorry to bother. Would you have any ideas how I could resolve this?

DonLakeFlyer, 2021-11-07:
Sorry I have nothing to do with HereLink.

iamczar, 2021-11-07:
Ah, Sorry - Did not know. But thanks for the reply

iamczar, 2021-11-08:
Hi @bugobliterator Sid. Would you know how I could resolve it or where I can look at?

meee1, 2021-11-09:
https://user-images.githubusercontent.com/18098277/140604935-5965d55c-ed2a-4752-a8c0-45dd20cca383.png
update your sdk version

iamczar, 2021-11-09:
Hi @meee1 Thanks for the reply.

I was trying to build the master with Qt 5.11.0 with java-8-openjdk and ndk version 15

to which sdk version I should Update? Latest?

Also I can see that the herelink-4.0.8 is more active than master

Should I be building that instead?

iamczar, 2021-11-09:
I tried building the herelink-4.0.8 - here is a video to give a bit of context -https://youtu.be/BFgIBO7sgM0

iamczar, 2021-11-09:
I went to compare the herelink-4.0.8 with against the current qgc main 

I find that there was no VideoManager on the herelink...

[image]

iamczar, 2021-11-24:
I was able to resolve this changing the version on the android manifest file

closed 2021-11-24 by iamczar