CubePilot/qgroundcontrol-herelink #27: GStreamer encountered a general supporting library error

karl-eaves, 2021-12-29:
**Ground Station**
Android 7 (Herelink)

**QGroundControl Version**
herelink-v4.0.8

**AutoPilot Board/Firmware**
Cube Orange - Ardupilot

**Build Computer**
Ubuntu 18

**QT Version**
5.11.0

**Android Build Tools**
30.0.0

**Android Platform**
android-25

**Android ABI**
armebai-v7a

**Android NDK**
r15c

**Android Build Tool**
android-g++

I am attempting to build a custom ground control application to be used on the Herelink. I have followed instructions from the QGC website for building QGC for Android in QT. I have also created the gstreamer-1.0-android-universal-1.14.4 directory within my main directory and extracted the gstreamer files here. I get no build errors/warnings regarding gstreamer when I build the app. However, the video stream does not work and I receive the following error in the console when attempting to view a video stream:

`
E QGroundControl: /project/source/src/VideoStreaming/VideoReceiver.cc:617 (static gboolean VideoReceiver::_onBusMessage(GstBus*, GstMessage*, gpointer)): GStreamer encountered a general supporting library error.`
`

If I set the gstreamer debug level to 5, I get the following extended debug output:

E QGroundControl: /project/source/src/VideoStreaming/VideoReceiver.cc:617 (static gboolean VideoReceiver::_onBusMessage(GstBus*, GstMessage*, gpointer)): GStreamer encountered a general supporting library error.
12-29 11:46:07.873   455  4458 W QGroundControl: gstvideodecoder.c:732 (gst_video_decoder_setcaps): GstreamerLog: <amcviddec-omxarmvideov5xxdecoder> Subclass refused caps
12-29 11:46:07.881   455  4458 E QGroundControl: gstamcvideodec.c:1962 (gst_amc_video_dec_set_format): GstreamerLog: <amcviddec-omxarmvideov5xxdecoder> Failed to negotiate
12-29 11:46:07.884   455  4458 W QGroundControl: gstamcvideodec.c:2033 (gst_amc_video_dec_set_format): GstreamerLog: <amcviddec-omxarmvideov5xxdecoder> error: Could not retreive application class loader
12-29 11:46:07.884   455  4458 W QGroundControl: gstvideodecoder.c:732 (gst_video_decoder_setcaps): GstreamerLog: <amcviddec-omxarmvideov5xxdecoder> Subclass refused caps
12-29 11:46:07.888   455  4467 W QGroundControl: gstrtspsrc.c:5995 (gst_rtspsrc_try_send): GstreamerLog: <rtsp-source> send interrupted
12-29 11:46:07.888   455  4467 W QGroundControl: gstrtspsrc.c:8242 (gst_rtspsrc_pause): GstreamerLog: <rtsp-source> PAUSE interrupted
12-29 11:46:07.894   455  4467 W QGroundControl: gstrtspsrc.c:5995 (gst_rtspsrc_try_send): GstreamerLog: <rtsp-source> send interrupted
12-29 11:46:07.895   455  4467 W QGroundControl: gstrtspsrc.c:7669 (gst_rtspsrc_close): GstreamerLog: <rtsp-source> TEARDOWN interrupted
12-29 11:46:08.125   455  4597 I QGroundControl: gstutils.c:3981 (gst_pad_create_stream_id_internal): GstreamerLog: <fakesrc129:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
12-29 11:46:08.460   455  4458 I QGroundControl: gstvideodecoder.c:933 (gst_video_decoder_drain_out): GstreamerLog: <amcviddec-omxarmvideov5xxdecoder> Sub-class should implement drain()
12-29 11:46:08.460   455  4458 I QGroundControl: gstbasesink.c:3145 (gst_base_sink_default_event): GstreamerLog: <qtsink1> stream-start event without group-id. Consider implementing group-id handling in the upstream elements
12-29 11:46:08.479   455  4458 W QGroundControl: gstpad.c:4226 (gst_pad_peer_query): GstreamerLog: <h264-parser:src> could not send sticky events
12-29 11:46:08.494   455  4458 W QGroundControl: gstamcvideodec.c:2033 (gst_amc_video_dec_set_format): GstreamerLog: <amcviddec-omxarmvideov5xxdecoder> error: Could not retreive application class loader
12-29 11:46:08.495   455  4458 E QGroundControl: /project/source/src/VideoStreaming/VideoReceiver.cc:617 (static gboolean VideoReceiver::_onBusMessage(GstBus*, GstMessage*, gpointer)): GStreamer encountered a general supporting library error.
`

Is there a library I am missing to make gstreamer work? From the guide it seems like the extracted gstreamer-1.0-android-universal-1.14.4 directory should include all the libraries needed for gstreamer.

 Do you have any recommendations for fixing / debugging this issue? I cannot find any information online about this. Again, I have followed the build instructions from the QGC Developers Guide, other than using android-g++ to build instead of clang because I cannot get the build to work with clang. 

Thanks

karl-eaves, 2021-12-29:
I have tried to install using this guide https://docs.cubepilot.org/user-guides/herelink/herelink-user-guides/oem-image-setup#example-app-change instead of directly installing with .apk. 

However, when I use this method, I get the following error:

`
12-29 14:52:50.427  2536  2553 E AndroidRuntime: FATAL EXCEPTION: qtMainLoopThread
12-29 14:52:50.427  2536  2553 E AndroidRuntime: Process: com.wisprsystems.wisprgroundcontrol, PID: 2536
12-29 14:52:50.427  2536  2553 E AndroidRuntime: java.lang.UnsatisfiedLinkError: dlopen failed: cannot locate symbol "OCSP_RESPID_free" referenced by "/oem/app/WisprGroundControl/lib/arm/libssl.so"...
12-29 14:52:50.427  2536  2553 E AndroidRuntime: 	at java.lang.Runtime.load0(Runtime.java:908)
12-29 14:52:50.427  2536  2553 E AndroidRuntime: 	at java.lang.System.load(System.java:1537)
12-29 14:52:50.427  2536  2553 E AndroidRuntime: 	at org.qtproject.qt5.android.QtNative$3.run(QtNative.java:217)
12-29 14:52:50.427  2536  2553 E AndroidRuntime: 	at org.qtproject.qt5.android.QtThread$2.run(QtThread.java:87)
12-29 14:52:50.427  2536  2553 E AndroidRuntime: 	at org.qtproject.qt5.android.QtThread$1.run(QtThread.java:61)
12-29 14:52:50.427  2536  2553 E AndroidRuntime: 	at java.lang.Thread.run(Thread.java:761)
12-29 14:52:50.429   732  2474 W ActivityManager:   Force finishing activity com.wisprsystems.wisprgroundcontrol/.QGCActivity

`

karl-eaves, 2021-12-29:
One thing to note: I had to comment out the line `nativeInit();` in QGCActivity.java. I just noticed that this is linked to the gst_android_init() function within main.cc. 

However, if I try to include this line, I receive the following error when running the app:

`
12-29 17:43:25.195 18151 18151 E AndroidRuntime: FATAL EXCEPTION: main
12-29 17:43:25.195 18151 18151 E AndroidRuntime: Process: com.wisprsystems.wisprgroundcontrol, PID: 18151
12-29 17:43:25.195 18151 18151 E AndroidRuntime: java.lang.UnsatisfiedLinkError: No implementation found for void com.wisprsystems.wisprgroundcontrol.QGCActivity.nativeInit() (tried Java_com_wisprsystems_wisprgroundcontrol_QGCActivity_nativeInit and Java_com_wisprsystems_wisprgroundcontrol_QGCActivity_nativeInit__)
12-29 17:43:25.195 18151 18151 E AndroidRuntime: 	at com.wisprsystems.wisprgroundcontrol.QGCActivity.nativeInit(Native Method)
12-29 17:43:25.195 18151 18151 E AndroidRuntime: 	at com.wisprsystems.wisprgroundcontrol.QGCActivity.onCreate(QGCActivity.java:206)
12-29 17:43:25.195 18151 18151 E AndroidRuntime: 	at android.app.Activity.performCreate(Activity.java:6682)
12-29 17:43:25.195 18151 18151 E AndroidRuntime: 	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1118)
12-29 17:43:25.195 18151 18151 E AndroidRuntime: 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2619)
12-29 17:43:25.195 18151 18151 E AndroidRuntime: 	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2727)
12-29 17:43:25.195 18151 18151 E AndroidRuntime: 	at android.app.ActivityThread.-wrap12(ActivityThread.java)
12-29 17:43:25.195 18151 18151 E AndroidRuntime: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1478)
12-29 17:43:25.195 18151 18151 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:102)
12-29 17:43:25.195 18151 18151 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:154)
12-29 17:43:25.195 18151 18151 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:6121)
12-29 17:43:25.195 18151 18151 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
12-29 17:43:25.195 18151 18151 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:889)
12-29 17:43:25.195 18151 18151 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:779)
12-29 17:43:25.198   735  8671 W ActivityManager:   Force finishing activity com.wisprsystems.wisprgroundcontrol/.QGCActivity
`

This could be the source of my problem. Do you have any insight on why the setNativeMethods() function is not being called within main.cc?

karl-eaves, 2021-12-30:
Upon digging a little more into this issue, it appears that the JNI_OnLoad() method within main.cc is never being called, which leads to setNativeMethods() not getting called, leading to the nativeInit() function not being implemented. This results in the above error `No implementation found for void org.mavlink.qgroundcontrol.QGCActivity.nativeInit()`. 

I have tried moving the setNativeMethods() function call to the top of the main() function, but I still got the same error `No implementation found for void org.mavlink.qgroundcontrol.QGCActivity.nativeInit()`. I'm guessing this is because the QGCActivity.java onCreate() function is being called before the main.cc main() function.

@meee1 @bugobliterator   Do you have any insight on what may be causing this JNI_OnLoad() function to not be called? Do you have any tips on how I should debug this issue?

iamczar, 2022-01-14:
@karl-eaves did you get further with this?, I am also having issues 

[image]

Ground Station
Android 7 (Herelink)

QGroundControl Version
herelink-v4.0.8

Build Computer
Ubuntu 20.04

QT Version - (versions i have tried)
5.11.0
5.12.6
5.15.2

Android Build Tools
30.0.0

Android Platform
android-25

Android ABI
armebai-v7a

Android NDK
r15c

Android Build Tool
android-g++

karl-eaves, 2022-01-14:
@iamczar I was able to get it to build by using the environment detailed here: https://github.com/CubePilot/qgroundcontrol-herelink/actions/runs/1430391344/workflow . I'd recommend upgrading to NDK r20 and using android-clang instead of android-g++. 

Also, if you take a look at that specific error message, it is because the variable `QGC_GST_STREAMING` is not being set. This variable should be set within src/VideoStreaming/VideoStreaming.pri. If you're building for android armv7, the most likely cause for this is that the gstreamer-1.0-android-universal-1.14.4/armv7 cannot be found in your repository during the build process. Check out this README for more info on that: https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoReceiver/README.md#android

iamczar, 2022-01-14:
@karl-eaves thanks for the prompt reply - im going to try :)

iamczar, 2022-01-14:
@karl-eaves I was able to build and deploy the herelink-v4.0.8 branch 

as per your recomendation - 

I downloaded the gstreamer from here - https://gstreamer.freedesktop.org/data/pkg/android/1.14.4/

I used android-ndk-r20b

using Android Qt 5.12.5 Clang armeabi-v7a

karl-eaves, 2022-01-14:
@iamczar Awesome! I'm glad that worked for you. I'm going to go ahead and close this issue

closed 2022-01-14 by karl-eaves

iamczar, 2022-01-14:
thanks again :)

DrivenShadow, 2025-01-20:
I'm getting the same error "GStreamer encountered a general supporting library error" when running on the Herelink an APK that I compiled using:

- Qt 5.15.2
- the now-current Stable_V4.4.3-Herelink at commit 6a32da0
- gstreamer-1.0-android-universal-1.18.6

Using the official 4.4.3 and 4.0.8 the video is shown as expected.

The object that throws the error is amcvideodec-omxarmvideov5xxdecoder0.

Do you have any suggestion on how to debug this error?

Comparing the contents of the APKs I see that some Qt libraries are different (for example libQt5Multimedia_arm64-v8a.so and
libQt5MultimediaQuick_arm64-v8a.so).