CubePilot/qgroundcontrol-herelink #56: 4.4 OTA Release

jonathan-k4, 2024-06-28:
I see that 4.4 was released about 3 weeks ago on this repo. However, I can't install it on my Herelink via ADB because the original 4.0.8 on there is part of the system apps. When will a system update be released for Herelink that includes the 4.4 version?

Thanks.

julianoes, 2024-06-29:
You should be able to sideload it using adb.

jagould2012, 2024-06-29:
You can’t sideload it. You receive a [INSTALL_FAILED_VERSION_DOWNGRADE] error. You also can’t delete the app. Presumably this is because it was signed as a system app during the herelink image build?

julianoes, 2024-07-03:
Ok, that's not ok then. I will look into this next week. I am confused because I have it side-loaded on my stock image with stock QGC 4.0.8.

jonathan-k4, 2024-07-03:
Is it possible you built it when you side loaded and built it with a different bundleId? That would get around the issue, would just be two separate apps. I was using the prebuilt apk from the 4.4 release page, which matches the one on the device.

Let me know if there is any adb output I can provide that will help you track it down when you get to looking at it. Thanks.

julianoes, 2024-07-03:
I'm pretty sure I tested the artifact from here: https://github.com/CubePilot/qgroundcontrol-herelink/releases/tag/v4.4.0-herelink
I'll try it again and double check.

julianoes, 2024-07-07:
Ok, for me v4.0.8 looks like this:

[image]
[image]

I then install v4.4.0 from [releases](https://github.com/CubePilot/qgroundcontrol-herelink/releases/tag/v4.4.0-herelink) using adb:

```
adb install QGroundControl-Herelink.apk
``` 

And I now have both versions installed:

[image]
[image]

Can you try to uninstall Herelink-QGroundControl first, and then try again?

And also paste a screenshot of the applications list?

jonathan-k4, 2024-07-12:
Steps to reproduce:

* Downloaded apk from https://github.com/CubePilot/qgroundcontrol-herelink/releases/download/v4.4.0-herelink/QGroundControl-Herelink.apk

* `adb install QGroundControl-Herelink.apk`

Error: 

```
adb: failed to install QGroundControl-Herelink.apk: Failure [INSTALL_FAILED_VERSION_DOWNGRADE]
```

* also attempted `adb install -r QGroundControl-Herelink.apk`

* `pm list packages -f | grep mavlink` lists 

```
package:/system/app/QGroundControl/QGroundControl.apk=org.mavlink.qgroundcontrol

```

which as I understand it, because its installed as a system app, I cant delete it or overwrite without root?

Screenshot of app and trying to delete it:

[image]

[image]

[image]

jonathan-k4, 2024-07-12:
Logcat output from when I try to delete:

```
07-12 09:22:29.678   744   772 W PackageManager: Attempt to delete unknown system package org.mavlink.qgroundcontrol
07-12 09:22:29.684   744   760 I ActivityManager: Force stopping org.mavlink.qgroundcontrol appid=10054 user=0: deletePackageX
07-12 09:22:29.786  1793  1793 D UninstallAppProgress: Uninstall failed for org.mavlink.qgroundcontrol with code -1
07-12 09:22:29.808   744   767 I ActivityManager: Displayed com.android.packageinstaller/.UninstallAppProgress: +187ms
```

jonathan-k4, 2024-07-12:
Some articles suggest disabling "verify apps over USB" to potentially force a way around this, but this option appears to have been disabled during the system image build:

[image]

julianoes, 2024-07-12:
Hang on, on your screenshot above I see two icons of QGC, so presumably you have two versions installed already? As far as my testing goes, the stock one does not need uninstalling.

jonathan-k4, 2024-07-12:
The second want ones a stock QGroundControl image from Android store. FYI - I ended up having to remove both to get adb install to work.

julianoes, 2024-07-12:
> That said, would recommend in future versions of the system image build that you maybe not load and sign the app as a system app?

Yes, that was not the goal. I'm still struggling to find out why it works for me. Let me check the IDs:

```
pm list packages -f | grep qgr                                                                                                       
package:/data/app/org.mavlink.qgroundcontrolbeta-2/base.apk=org.mavlink.qgroundcontrolbeta
package:/system/app/QGroundControl/QGroundControl.apk=org.mavlink.qgroundcontrol
```

Now I uninstall v4.4.0.1 and check again:

```
pm list packages -f | grep qgr                                                                                                       
package:/system/app/QGroundControl/QGroundControl.apk=org.mavlink.qgroundcontrol
```

julianoes, 2024-07-12:
Now I install it again.

```
pm list packages -f | grep qgr                                                                                                       
package:/data/app/org.mavlink.qgroundcontrolbeta-1/base.apk=org.mavlink.qgroundcontrolbeta
package:/system/app/QGroundControl/QGroundControl.apk=org.mavlink.qgroundcontrol
```

julianoes, 2024-07-12:
The beta name is set here, presumably. I'm not 100% sure how this all works.

https://github.com/CubePilot/qgroundcontrol-herelink/blob/Stable_V4.4-Herelink/tools/update_android_manifest_package.sh#L6

jonathan-k4, 2024-07-12:
The second app on mine also had the `org.mavlink.qgroundcontrolbeta` package name, so it wasn't part of any of the commands I listed above. It was also installed in the `data` directory like yours (user space) so was easily removed. The preloaded one was installed in the `system` directory, and I had to do the `pm uninstall -k --user 0 org.mavlink.qgroundcontrol` to remove it.

Hope that helps.

closed 2026-07-02 by julianoes